splitti/MuPiBox #6: Change sound output to headphone jack

oswaldo-be, 2022-05-28:
Hej there,

thank you very much for this project. I only have one phoniebox with the hifiberry.
The other one works with the headphone jack.
Can you give a short explanation please how I can change it?
I fiddled around with phoniebox for one year and have little experience in the config how this worked...

Thank you very much

nero0345, 2022-05-28:
Hello,

the question is whether you want to use only the headphone jack or the Hifiberry on the MuPiBox, or both. I have made after much testing the experience that both is really heavy and also the quality is not so good. But you can use both with the MuPiBox and depending on what you just want to have in the web interface easily select. Then the output is only output via dise output. The automatic change does not work with it. But what works are Bluetooth headphones and so the automatic switch works.
I hope I could help you.
If you have further questions, just come to our Discord and we can help you even more.

https://discord.gg/4EjCgpCbbe

nero0345

splitti, 2022-05-30:
In version 1.0.8 the jack is an option in the configuration.

closed 2022-05-30 by splitti