splitti/MuPiBox #11: Start up time verlängert sich kontinuierlich

passuff, 2022-07-04:
Eine frisch aufgesetzte Box mit importierten Einstellungen startet  bei mir in ca 16 Sekunden. Nach nun 3 Wochen sind daraus 46 Sekunden geworden, ohne dass eine Änderung vorgenommen wurde.

splitti, 2022-10-05:
Wie schaltest Du die Box ein oder aus?

passuff, 2022-10-05:
Mit dem on/off shim.

splitti, 2022-12-28:
Besteht das Problem noch immer?

passuff, 2022-12-29:
Nein, ist seit 2.04 nicht mehr aufgetreten.

Hunv, 2023-05-27:
Moin,
ich habe das Problem auch. Und zwar mit Version 3.0.1.
Der Boot mit eine Raspi 4 1GB benötigt
- bis zum Abspielen des Startsounds ca. 21 Sekunden
- bis ich die Seite mit den Hörbüchern und deren Artist icons (alles Spotify) sehe: ca. 34 Sekunden

Das war früher definitiv mal schneller und mein Gefühl über die letzten Monate ist, dass es immer länger wird.
Die MuPiBox ist seit Weihnachten 2022 im Betrieb.

nero0345, 2023-07-03:
Das Problem sollte schon seit längeren gelöst sein.

closed 2023-07-03 by nero0345