splitti/MuPiBox #13: Featurerequest wsdd support and popping prevention for I2S Soundcards

MarkusLange, 2022-09-18:
Hello,

the MuPiBox is really fine and works nice, there are 2 pionts I would have to add.
1.
Support for wsdd like [this](https://github.com/christgau/wsdd) so I can see this device inside my Windows Network, since the SMB V1 support is dropped by Winodws 10, works well on the Dietpi .
2.
Adding this Workaround that prevents the popping from I2S Soundcards (aplay.service)
```
[Unit]
Description=Invoke aplay from /dev/zero at system start.
After=pm2-dietpi.service

[Service]
ExecStart=/usr/bin/aplay -D default -t raw -r 44100 -c 2 -f S16_LE /dev/zero

[Install]
WantedBy=multi-user.target
```
borrowed [there](https://github.com/adafruit/Raspberry-Pi-Installer-Scripts/blob/a779800e50551c4ec48f3c7aae31a53b6011b9eb/i2samp.sh#L447), keeps the Audiocard online so there are no pop sound if the card is reactivated.

trewbene, 2022-10-05:
Especially when using the Adafruit MAX98357 I2S (https://learn.adafruit.com/adafruit-max98357-i2s-class-d-mono-amp/raspberry-pi-usage), it unfortunately cracks loudly when starting a sound file.
The mentioned workaround solves this problem. It would be great if this could be activated as an option in the settings, as this is not necessary for every i2s device (for example Hifiberry Mini Amp).
In my opinion, using the Max98357 has the advantage over the Hifiberry Mini Amp that mono output works without problems.

MarkusLange, 2022-10-05:
The Adafruit I2S 3W Stereo Speaker Bonnet with two MAX98357 has also the pop sound when start playing

MarkusLange, 2023-12-24:
Since Bookworm wsdd is part of debian and it can installed via apt install wsdd

splitti, 2024-01-15:
Sorry for the long waiting  time... wsdd will be implemented soon.
I will add an option, to keep the soundcard active to provide this sound.

splitti, 2024-01-19:
wsdd added to v3.3.0