splitti/MuPiBox #16: Featurerequest: Sleeptimer

tom3kcgn, 2022-09-24:
Ich möchte im Player einen Sleeptimer setzen können.
Optionen:
* 15 Minuten
* 30 Minuten
* 60 Minuten
* Am Ende des Tracks
* Am Ende des Albums / Hörspiel

hyperbit00, 2022-10-07:
Ich kann diesen requests nur unterstützen. Hatte bisher Probleme mit dem IdleTimer. Dieser hat (zumindest bis Dev 1.0.10) nicht immer zuverlässig die Box abgeschaltet.

Ein Sleeptimer kann daher evtl helfen.

Wurde die Zeit noch un einen eigenen Wert erweitern, damit eine Freie Laufzeit bestimmt werden kann .

splitti, 2022-12-11:
Der Idletimer ist gefixt. Ist denn ein genereller Timer, der nicht abhängig davon ist, ob gerade etwas gespielt wird oder nicht, weiterhin erwünscht?

tom3kcgn, 2022-12-11:
Trotz des idle Timer fände ich die Option Klasse.
Hat aber keine so hohe Prio.

splitti, 2022-12-16:
Wie stellt Ihr Euch den Timer denn vor? Aktivierung im Admin-Interface? Oder eine generelle Einstellung?

tom3kcgn, 2022-12-16:
> Wie stellt Ihr Euch den Timer denn vor? Aktivierung im Admin-Interface? Oder eine generelle Einstellung?

tom3kcgn, 2022-12-16:
Ich denke im Adminbereich reicht das aus, da die Kids das selbständig nicht können (jedenfalls die kleineren nicht) und der Knopf sonst nur zur Verwirrung führt.

hyperbit00, 2022-12-16:
Tatsächlich dort wo auch der Idletimer eingestellt wird =) wird.

#Offtopic ein + & - bei der Lautstärkeeinstellung (in den Einstellungen) wäre auch fein. Auf einem Smartphone ist dieser Sliderbalken nur schwer zu bedienen (Entschuldigung fürs #offtopic)

splitti, 2022-12-28:
Wird im kommenden Release berücksichtigt!

closed 2022-12-28 by splitti