splitti/MuPiBox #28: Incorrect / old TTS files after changing TTS Language

rgrollfitz, 2023-08-15:
Initially the TTS language is set to English. If, before changing this default, one of the titles is read out and then the language is changed, the old TTS audio file will be used, since this is stored locally - without an identifier for the language used. _(/home/dietpi/MuPiBox/tts_files/)_

Presumably the TTS files are stored locally to avoid having to access an interface again (which most likely only allows limited access) when playing repeatedly.

Maybe it would be possible to attach an identifier to the TTS file, which is updated as soon as this language identifier differs from the currently set TTS language.

splitti, 2023-12-22:
Fixed in new versions

closed 2023-12-22 by splitti

MarkusLange, 2025-01-08:
Looks like it's back in 4.2.1