splitti/MuPiBox #29: Fresh installation fails

fseifert, 2023-08-27:
I want to reset one of my mupiboxes and did a fresh install using the links from mupibox.de.

I downloaded the current dietpi bullseye release (which dates to 31st of july) along with the current dietpi startup script.
The install did not show any issues (at least i didn't notice any) and the box starts, with the screen with touch is available and the webui is functional.
But the screen only shows that the backend is not running and, indeed, on port 8200 is nothing present.

Checking the install script, i found the install for the backend which requires npm.
So i tested manually installing npm, but it consistently fails with the following error:
> root@phoniebox:~# apt-get install npm
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:
The following packages have unmet dependencies:
 libnode72 : Conflicts: nodejs-legacy
 nodejs : Conflicts: npm
E: Unable to correct problems, you have held broken packages.

Nexodaru, 2023-09-01:
I have the same Problem. After the first start chromium shows: "This site cant be reached". 
I checked the logs in /boot/autosetup.log and found the same lines as the Issue-Creator. 

`## Run `sudo apt-get install -y nodejs` to install Node.js 16.x and npm
## You may also need development tools to build native addons:
     sudo apt-get install gcc g++ make
## To install the Yarn package manager, run:
     curl -sL https://dl.yarnpkg.com/debian/pubkey.gpg | gpg --dearmor | sudo tee /usr/share/keyrings/yarnkey.gpg >/dev>     echo "deb [signed-by=/usr/share/keyrings/yarnkey.gpg] https://dl.yarnpkg.com/debian stable main" | sudo tee /etc/a>     sudo apt-get update && sudo apt-get install yarn

Paketlisten werden gelesen…
Abhängigkeitsbaum wird aufgebaut…
Statusinformationen werden eingelesen…
Einige Pakete konnten nicht installiert werden. Das kann bedeuten, dass
Sie eine unmögliche Situation angefordert haben oder, wenn Sie die
Unstable-Distribution verwenden, dass einige erforderliche Pakete noch
nicht erstellt wurden oder Incoming noch nicht verlassen haben.
Die folgenden Informationen helfen Ihnen vielleicht, die Situation zu lösen:

Die folgenden Pakete haben unerfüllte Abhängigkeiten:
 libnode72 : Kollidiert mit: nodejs-legacy
 nodejs : Kollidiert mit: npm
E: Probleme können nicht korrigiert werden, Sie haben zurückgehaltene defekte Pakete.
sudo: npm: Befehl nicht gefunden
sudo: npm: Befehl nicht gefunden
bash: Zeile 63: pm2: Kommando nicht gefunden.`

Nexodaru, 2023-09-01:
At first i tried only to change these line
```
curl -sL https://deb.nodesource.com/setup_16.x | sudo -E bash - >&3 2>&3
```
to
```
sudo apt-get install -y ca-certificates curl gnupg >&3 2>&3
sudo mkdir -p /etc/apt/keyrings >&3 2>&3
curl -fsSL https://deb.nodesource.com/gpgkey/nodesource-repo.gpg.key | sudo gpg --dearmor -o /etc/apt/keyrings/nodesource.gpg >&3 2>&3

NODE_MAJOR=16
echo "deb [signed-by=/etc/apt/keyrings/nodesource.gpg] https://deb.nodesource.com/node_$NODE_MAJOR.x nodistro main" | sudo tee /etc/apt/sources.list.d/nodesource.list >&3 2>&3

sudo apt-get update >&3 2>&3
```
Then i placed the script on the root path of a fresh created dietPi-SD-Card and renamed it to "Automation_Custom_Script.sh".
To prevent the faulty script running i comment out the following line in "diet-pi.txt"

`AUTO_SETUP_CUSTOM_SCRIPT_EXEC=https://mupibox.de/version/latest/autosetup/autostart.sh`

Sadly the problem the problem persisted. 

In the end, the problem is probably these lines of code, right after adding the NodeJs repo:
https://github.com/splitti/MuPiBox/blob/c8393f211d135ef9d4f9864393ec1360be303b9a/autosetup/autosetup.sh#L57
```
	if [ `getconf LONG_BIT` != 32 ]; then
		sudo apt-get install -y nodejs >&3 2>&3
	else
		sudo apt-get install -y nodejs npm >&3 2>&3
	fi
```
Following the instructions on the website, I used the 32 bit image from dietPi, which ends up in the else branch of the if condition. Here we try to install NodeJs and Npm. However, as it seems, this is not necessary and also not possible, since npm is already installed with nodeJs.
I installed nodejs manually with the command
`sudo apt-get install -y nodejs' 
 and then ran the remaining lines of the script. Now my first MuPiBox starts up to the WebUI.

During the installation I noticed that the package mplayer wrapper removed support for NodeJs 16. 
(See https://github.com/derhuerst/mplayer-wrapper/commit/2a66e101800ed8b6ece6db8015e3b1c7679c0c10)
However, the installation seems successfull. I hope the Mupibox is still stable.

fseifert, 2023-10-30:
A few days ago i tried another fresh install and its seems the version 3.1.3 has fixed the install issue.

I noticed just one issue (don't know why i didn't stumble across this with all the installs before): I set the new hostname to mupibox04 and dns and dhcp just works fine, just the display didn't show anything useful. I first thought it was again the issue i described above, but it was much easier this time: The mupiboxconfig.json in /etc still had the entry "host": "MuPiBox". After changing this to mupibox04 and a reboot everything was fine.

I don't know, if this replacement is broken or if i changed the hostname all the times later.

Maybe this hint helps other and regarding the initial issue, as said, 3.1.3 has solved it. Thanks.

closed 2023-10-30 by fseifert