splitti/MuPiBox #30: Media selection stops working

fseifert, 2023-08-27:
I had two mupiboxes running for some time without an issue. Both were updated some weeks(?) ago to 3.1.2 and worked without any issues.

Now some days ago, one of the two boxes, i am unable to select any music. The screen shows all media and the touchscreen itself is fully functional, i.e. i can scroll, change category, enter new wlan credententials. But pointing at any media, just nothing happens. Also pointing at the title, no TTS does happen. Strangely, the same applies to the browser application from my pc. I can scroll, but clicking on any media, just nothing happens. The second box works with now flaws.

RobertoWa, 2023-08-27:
i can confirm this issue too. a reboot solves that.

fseifert, 2023-08-27:
Unfortunately, a reboot does not fix the issue for me, the issue is persistent.

nero0345, 2023-08-28:
Over SSH on the Pi and make this comment.

`sudo wget https://mupibox.de/version/latest/config/templates/monitor.json -O ~/.mupibox/Sonos-Kids-Controller-master/server/config/monitor.json`

And after this you make a update to the dev version in the webinterface.

fseifert, 2023-08-28:
That did it. The monitor.json was empty. (The other box i have had Off as value.) Filling the content brings the box back to live. Thanks.

closed 2023-08-28 by fseifert