splitti/MuPiBox #35: Spotify Suchsyntax

hollywoot, 2023-12-26:
das Hinzufügen von Titeln mithilfe der Spotify-Query ist nicht mehr möglich.
Aufgefallen ist es, weil plötzlich einige Titel nicht mehr auf der Box waren.

`artist:"Das Original-Hörspiel zum Disney"` hatte bis vor einiger Zeit wunderbar die ganzen Hörspiele der Disneyfilme gefiltert.
Die gleiche Suche liefert mittlerweile keinerlei Ergebnis mehr.

Das Beispiel auf der Homepage funktioniert auch nicht.
`artist:“LEGO Ninjago“ AND album:“Folge“`

Weiteres Beispiel:
`artist:"PJ Masks" AND album:"Folge"` liefert ebenso nichts mehr.

ronbal, 2023-12-26:
Probiere es mal ohne Anführungszeichen also artist:Dragons die jungen Drachenretter z.b. wichtig zwischen artist: und dem suchstring kein Leerzeichen setzen

hollywoot, 2023-12-26:
Dein Beispiel funktioniert.
Es funktioniert auch `artist:"PJ Masks"`.
Aber versuche mal bestimmte Sachen mittels `AND` rauszufiltern. Das will nicht mehr. 
Ich will bei den Disney-Hörspielen z.B. die "Real"-Filme nicht mit im Ordner haben. Also sollen sie rausgefiltert werden.
Auch heißen einige Hörspiele "Das Original-Hörspiel zum Disney Film", andere heißen "[...] zum Disney/Pixar Film". Aus diesem Grund hatte ich "[...] zum Disney" gewählt. Das hat bislang auch ohne Probleme funktioniert. 
Irgendwann hatte sich mein Sohn beschwert, dass die Hörspiele nicht mehr da sind. 
Ich hatte keine Änderungen gemacht - es hat einfach aufgehört zu funktionieren.

Ggf. hat es ja Änderungen seitens Spotify gegeben?

ronbal, 2023-12-26:
Ja spotify hat die api geändert and gibt es wohl nicht mehr nur + und - so habe ich es verstanden. Binaber auch nich nicht genau dahintergestiegen

nero0345, 2023-12-26:
Wie mein Vorredner geschrieben hat das AND gibt es nicht mehr und wird durch ein + ersetzt.

closed 2023-12-26 by nero0345

hollywoot, 2023-12-26:
Gut zu wissen.
Hab mal einige Varianten ausprobiert. 

Im Falle von den Disney-Hörspielen .... keine Chance.
Das war "früher" echt einfach. So muss ich alle Hörspiele einzeln hinzufügen.

Vielleicht kriegt es ja jemand anders hin eine funktionierende Query zu basteln