splitti/MuPiBox #41: site cant be reached

hyperbit00, 2024-01-23:
Ich habe ein Problem, welches nur sporadisch auftritt.

Nach dem Starten kommt die weiße Cromebrowser-Seite "This Site can't be reached" - diese verschwindet nach einiger Zeit wieder und das normale Userinterface (Mupibox) erscheint dann.

Zu erst ist dieser Fehler nach einer kompletten Neuinstallation aufgetreten. Nach diversen Neustarts und einigen Änderungen/Settings im Webinterface erscheint der Fehler aktuell nicht mehr

splitti, 2024-01-31:
Kannst du irgendwelche Unterschiede ausmachen, wenn das Problem auftritt? Z.B. kein Wifi Empfang?

hyperbit00, 2024-02-01:
Also das Webinterface via PC war immer erreichbar. Das ganze passiert mit externen WLan Stick oder Ohne.

Ebenfalls wurde der Raspberry Pi 3b durch einen 4b bereits getauscht. Ohne Erfolg.

hyperbit00, 2024-02-01:
Passiert ebenfalls in V7 wie n V8

splitti, 2024-02-14:
Bitte beachte, dass avoid_warnings die CPU nicht drosselt, sondern nur die Warnungen nicht mehr anzeigt. Wenn Du undervoltage hast, kann das zu Problemen führen. Hier stellt sich auch die Frage, was hast Du für eine microSD Karte? Ist Swap an?

hyperbit00, 2024-02-14:
USB Stick und kein swaping . Im Discord hatte jemand letztens das gleiche Problem. 

Bit zwischenzeitlich auf nen PI 4 gewechselt das Problem tritt nun  nur noch sporadisch auf und dann nur wenige augenblicke

htom89, 2024-02-15:
Hi, bei mir tritt das Problem auf seit ich den OnOFF SHIM verbaut habe. 
Nach Disable Warnings (Throtteling Warning) und update beta und wieder zurück auf 3.3.3. kommt das Problem auch nur noch kurz beim Start. Dafür bei jedem.

oswaldo-be, 2024-03-10:
Bei mir tritt der Fehler jetzt nach Update auf die 3.3.4 beta auf. Hostname auf mupibox zurück und wieder auf den eigentlichen ändern bringt auch nix.
Zum Glück erreiche ich die GUI per Browser immer über den jeweils eingetragenen Host.
Ich versuche auf die stable zurück zu wechseln und hoffe....

oswaldo-be, 2024-03-17:
Ich bin auf bookworm und habe gerade das Update auf stable 4.0.4 durchgeführt.
Ich komme unter meiner festgelegten Adresse zwar auf die admin GUI, aber an der box nur This site can't be reached.
Habe den host-namen mal auf mupibox zurück gesetzt, auch das half nix.

oswaldo-be, 2024-11-12:
Also den Fehler hab ich seit nach 4.0.4 nicht mehr gesehen :-)