splitti/MuPiBox #45: No Spotify playback after long idle

passuff, 2024-02-16:
There two issues with Spotify playback independent from used Mupibox version. 

1. Start Mupi,  play Spotify track, stop, let Mupibox untouched for several hours, try to play again - > no playback

After a restart, playback is working flawless.

2. Do not use Spotify for a week (maybe less). Start Mupibox, try to play a Spotify track- > no playback

Only a new Spotify config setup will bring Spotify back to work. 

Logs can be provided...

passuff, 2024-02-27:
EDIT for issue 2:
THe issue can't be solved by renewing the spotify config. The only way is to use spotify on another device and chose Mupibox as speaker. After that, Spotify playback on Mupibox is restored. 
Logs can be provided if necessary.

christianerbsmehl, 2024-04-06:
I'm experiencing the same problem. Spotify playback after pausing about an hour is not possible. I have to restart the box to get it working again.

ronbal, 2024-08-18:
> EDIT for issue 2:
> THe issue can't be solved by renewing the spotify config. The only way is to use spotify on another device and chose Mupibox as speaker. After that, Spotify playback on Mupibox is restored. 
> Logs can be provided if necessary.

Same herein in one box