splitti/MuPiBox #50: Spotify stops working after some days not using Spotify [bug]

passuff, 2024-04-26:
If you dont use spotify for more than some days (a week?), spotify playback is not working.
The only way to recover spotify playback is to use spotify on another device and chose Mupibox as speaker. After that, Spotify playback on Mupibox is restored.
Logs can be provided if necessary.

AndroidFactoryDE, 2024-06-01:
I have observed the same issue with version 4.0.4

Also the proposed workaround to recover Spotify has worked for me.

pfasante, 2024-08-18:
seems like a duplicate of #45

passuff, 2024-08-18:
No. Its different. One is idle (powered box but no usage for 6h) and the
other topic is for not powered box (1 week no Spotify usage).

RobertoWa, 2024-08-18:
i am facing this issue too.

mupibox is not able to play as a standalone client anymore. it happened from one day to the next.

settings interface under spotify step 4 no longer works smoothly.

fifafu, 2024-11-03:
I have also encountered this a lot, even with the latest dev versions. 

Is this maybe because of the setActiveDevice function ? https://github.com/splitti/MuPiBox/blob/84fad9f276535656b313b5ff1db54a6ce157a3a0/bin/nodejs/spotify-control.js#L329

This seems to randomly set the active device to the first device returned by the Spotify API. However I haven't looked deep enough into the code to understand when this is called. For testing I have now overridden this:

```
      activeDevice = config.spotify.deviceId;
```

to always assign the device id of my mupibox instead. I'll keep monitoring whether that helps...

nero0345, 2024-11-12:
Does the problem still occur in the current dev version?

passuff, 2024-11-19:
Yes. Just occurred again on DEV 2024-11-15.