splitti/MuPiBox #53: Add media, Spotify URL, error "The id is not valide or you have no internet connection!"

MatzeBeck, 2024-06-01:
When I try to add media data with "Source type" "Spotify URL" I always get the error message "The id is not valide or you have no internet connection!" 

I tried different links (artits, albums, ...), also remove the intl-de from the link.

One example:
https://open.spotify.com/intl-de/artist/6b9DQDp0xKeYM2vdK9pAE6

Adding Spotify with Search Query is no problem and the box works perfectly with Search Query.

Where could the problem be?
Thank you!

nero0345, 2024-06-02:
Which version do you have? Are you using IP or host name for the web interface? If IP, then please search for IP Control in the web interface under Admin and activate it.

MatzeBeck, 2024-06-03:
Use the IP Control solved the problem.
Thank you for the quick support and the great Software!

I use Version 4.0.5 and also tested 4.0.4