splitti/MuPiBox #54: Feature Request: Key lock for the software buttons on the display

Lhotze, 2024-06-02:
A locking function for display operation either via a decimated software button or, for example, via web interface / Telegram / MQTT. Definitely desirable for small children, discordant siblings or curious visitors in order to be able to enjoy a running playlist/album/song to the end without interruption. Hopefully, constantly jumping up, screaming in fright, nervous breakdowns or regular adjustments will be a thing of the past then.

Greetz

brio1009, 2024-10-16:
Hi, nice suggestion. I would keep it on the box only, meaning locking and unlocking works directly on the box. Would you have a suggestion on how unlocking could be realized in this case? Would a "swipe to unlock" be sufficient?

Lhotze, 2024-10-18:
I came up with the idea because my two-year-old regularly interrupts the music flow of her two siblings.

So my suggestion is primarily about a function that prevents this and not about locking the box for use by everyone except the admin interface user.

Accordingly, a swipe to unlock implementation would probably be ideal. The little one could no longer change anything, the big ones could still do it and if the little one does swipe around on the box, there is still enough reaction time.
Topp