splitti/MuPiBox #55: Spotify login not working, deprecated username/password authentification

pfasante, 2024-08-18:
As discussed in discord, seems to be based on https://github.com/librespot-org/librespot/issues/1308

splitti, 2024-08-19:
Here is a possible workaround:
https://github.com/Spotifyd/spotifyd/issues/1293#issuecomment-2296143564

gladwig, 2024-08-25:
Is the workaround working for anyone with Mupibox?

I was able to create the credentials.json and got spotfiyd to run by removing the username/password from spotifyd.conf manually. Mupibox does show up as a device in the Spotify app and I am able to play on Mupibox from the Spotify app on my PC.

I can then select Mupibox as the active device in the Mupibox admin interface, but I am not able to play songs from the Mupibox itself. It only works when I select Mupibox from the Spotify app as the the active device. But this does not survive a reboot, making it useless.

In .pm2/logs/spotify-control-error.log I get these errors:
```WebapiPlayerError: An error occurred while communicating with Spotify's Web API.
Details: Player command failed: No active device found NO_ACTIVE_DEVICE.
    at _toError (/home/dietpi/.mupibox/spotifycontroller-main/node_modules/spotify-web-api-node/src/http-manager.js:35:12)
    at /home/dietpi/.mupibox/spotifycontroller-main/node_modules/spotify-web-api-node/src/http-manager.js:71:25
    at Request.callback (/home/dietpi/.mupibox/spotifycontroller-main/node_modules/superagent/lib/node/index.js:905:3)
    at /home/dietpi/.mupibox/spotifycontroller-main/node_modules/superagent/lib/node/index.js:1127:20
    at IncomingMessage.<anonymous> (/home/dietpi/.mupibox/spotifycontroller-main/node_modules/superagent/lib/node/parsers/json.js:22:7)
    at Stream.emit (node:events:517:28)
    at Unzip.<anonymous> (/home/dietpi/.mupibox/spotifycontroller-main/node_modules/superagent/lib/node/unzip.js:53:12)
    at Unzip.emit (node:events:517:28)
    at endReadableNT (node:internal/streams/readable:1400:12)
    at process.processTicksAndRejections (node:internal/process/task_queues:82:21) {
  body: {
    error: {
      status: 404,
      message: 'Player command failed: No active device found',
      reason: 'NO_ACTIVE_DEVICE'
    }
  },```

gladwig, 2024-08-25:
Ok, disregard that. The problem was the missing username, with username_cmd in the spotifyd.conf it works now.

Also saw that there is a new release, I will try it out.

3052, 2024-08-31:
> deprecated username/password authentification 

its not. the current Spotify Android client definitely still support email/password, Ive been testing it all day. in fact my code from last month still works as is, if you add a user-agent

brio1009, 2024-10-16:
The username/password authentication has been removed with the 4.1.X releases of MupiBox. There were still some lingering issues that are either already fixed in "dev" or will be fixed with #66 . Therefore closing this issue.

closed 2024-10-16 by brio1009