splitti/MuPiBox #72: Volume Startup and Shutdown Sound

MarkusLange, 2024-10-22:
Startup and Shutdown Sound are orientated on the Sound Value, the problem is if the last mp3 was heared on a high volume the Shutdown Sound is realy loud, a fixed value set in the options would be good so there is no loud Shutdown if the Kid hears a quiet mp3 on max volume

fade2metal, 2025-01-13:
I also have this issue. I already changed the volume values in the script files off_trigger.sh from /usr/bin/pactl set-sink-volume @DEFAULT_SINK@ ${START_VOLUME}%  to /usr/bin/pactl set-sink-volume @DEFAULT_SINK@ 10%  and the same in shutdown_sound.sh but when I reboot the Rpi with sudo reboot the sound ist extremely loud. And when the timer that can be configured in UI ist at 0 and the Pi shuts down it's extremely loud too! So when my daughter listens to an audiobook and the timer runs to 0 she ist awake again because the sound ist much louder than the Audiobook she was listening to.
@splitti: Is the shutdown sound in any system script and runs automatically?

MarkusLange, 2025-01-16:
As far as I have discover the volume of the shutdown.sound has a higher basevolume than the startup.sound so there is one point you can shift it a bit. Next is seperate startup and shutdown volume so I can choose by myself and can set shutdown to zero

b-nobis, 2025-01-16:
The commit 88db8ef1db65cdde723f41bc9e762aac9d8648ff fixed the loud shutdown sound issue for `sudo reboot` for me.