splitti/MuPiBox #73: Add Spotify Media via URL not possible - Validation error

calibrationcube, 2024-10-28:
It is no longer possible to add Spotify content via Spotify URL.
The message appears: URL is invalid or No internet connection

Notes:
Adding via Spotify Search Syntax works. 
Adding an ID manually via data.json also works without any problems.
Play spotify Songs works (so no connection error)

Currently tested version: DEV 2024-10-18

fifafu, 2024-11-02:
Same issue here with DEV 2024-11-01. Thanks for the workarounds @calibrationcube !

fifafu, 2024-11-02:
In my case this happens because it tries to connect to http://mupibox:5005/validate  (even if I set ip_control_backend to true)
For some reason that doesn't work here. I'm not sure whether it's an issue with my network setup or whether the spotify service doesn't listen on the hostname. I changed my mupibox host from mupibox to just the ip address of the raspberry, and now it's working fine again.

calibrationcube, 2024-11-04:
Can confirm, that setting the Hostname to IP also works for me!
-> Since IP = Hostname no validation error

closed 2024-11-10 by nero0345