splitti/MuPiBox #102: Feature Request local files cover picutre

Lippsson, 2024-11-16:
When loading local files. 
Change the way to look up cover pictures

1. Mupibox software should with every song/file lookup if there is a cover picture saved in the ID3-tag information and use this. 
2. If no cover is saved, check if there is a jpg file with the same name as the musicfile and use this. 
3. If (1) and (2) is not, check if there is any picture in the folder (cover.jpg) and use this   (STEP 3 is already integrated in Mupibox software)
4. If (1-3) is not, use Artistcover or gerneral Mupibox logo. (STEP 4 is already integrated in Mupibox software)

gnorbsl, 2024-12-18:
This will require ffmpeg to be installed on the MuPibox, my local version does this but I have not created a PR for it yet. I also wrote a script that generates this cover and also renames all files and folders based on the structure MuPibox expects

```bash 
#!/bin/bash

# First organize and rename files
find . -type f -name "*.mp3" | while read file; do
    album=$(eyeD3 "$file" | grep "album:" | cut -d":" -f2- | sed 's/^[ ]*//' | sed 's/[ ]*$//' | tr -cd '[:alnum:] .-')
    title=$(eyeD3 "$file" | grep "title:" | cut -d":" -f2- | sed 's/^[ ]*//' | sed 's/[ ]*$//' | tr -cd '[:alnum:] .-')
    
    if [ ! -z "$album" ] && [ ! -z "$title" ]; then
        mkdir -p "./$album"
        
        new_filename="${title}.mp3"
        
        mv "$file" "./$album/$new_filename"
        echo "Moved '$file' to '$album/$new_filename'"
        
        # Extract cover art if it doesn't already exist
        if [ ! -f "./$album/cover.jpg" ]; then
            ffmpeg -i "./$album/$new_filename" -an -vcodec copy "./$album/cover.jpg" 2>/dev/null
            if [ -f "./$album/cover.jpg" ]; then
                echo "Extracted cover art for '$album'"
            fi
        fi
    else
        echo "Missing album or title tag for '$file'"
    fi
done

# Remove all empty directories
find . -type d -empty -delete
echo "Removed empty directories"```

just copy this into a cover.sh file do `chmod +x cover.sh ` and then you can just upload it to the box and you have a cover

Lippsson, 2025-01-11:
Hello @gnorbsl!

Thanks for your reply. I think you misunderstood me. I have all cover files already. I have them also integraded in the mp3 file with mp3tag. Here an example: 
[image]
Now when the mupibox open the album with the 2 files should the mupibox in the first place look after the cover who are integrated in the mp3 files. if there are no cover there it should search for the regular jpg file int the album folder.